CIR-KIT/fifth_robot_pkg #1: 機体製作の進捗や目標

forno, 2016-09-14:
来週末まで機体製作強化週間的なノリで機体製作を進めていきます。
機体製作に関わるログ残し的な意味と、現状の把握と、メモを兼ねてハードウェア用のIssueを開きます。

とりあえずの目標は
機体製作を来週末の9/24まで
借り組み(とりあえず動く)まで進めましょう

HajimeTokuyasu, 2016-09-15:
・ 部品の注文が完了しました
 ・ 早ければ 21 日の水曜日に届きます.

 → 2~3日で組み立てを完了し, 点検, 導通確認など完了しておきましょう.

HajimeTokuyasu, 2016-09-15:
Kinetic での buildに成功. 依存関係追加する.

forno, 2016-09-15:
電装についてのIssue https://github.com/CIR-KIT/fifth_robot_pkg/issues/6 を開きました。
電気・電子の関係の話はそちらでお願いします。

yasu80, 2016-09-23:
部品は届きましたが, キャスタ(従輪)他いくつかの部品がありませんので, 今日組み立てられるところまで組み立てておきます
ソフトの稼働確認まではできそうです.

yasu80, 2016-09-23:
さきほど かりくみたてが かんりょうしました.
しかしながら 導通 絶縁 などの確認がまだです,
これらの作業について風呂ーを作っておきます.

forno, 2016-09-23:
頑張れ、しかしその前に寝ろ。0 a.m. ~ 8 a.m.の間に仕事せんでもええんやで。

yasu80, 2016-09-23:
ごめんなさい, でも組み立ては一人でやり遂げられましたぜ !!
[image]

yasu80, 2016-10-02:
・ 巻き込み防止策
 ・ 高さ稼ぎ策
 ・ 防雨策
 ・ 緊急停止スイッチ実装
etcをやってもらいたいんですが, この辺は一年生に協力を仰ぎやす分野だと思うんで明日そうします

forno, 2016-10-15:
各作業が独立して稼働できる状態になってきたと思うので、ジェネラルすぎるこのIssueの存在価値がなくなってきたかと思います。
このIssueは閉じて、具体的なIssueを次々開いていくことをおすすめします。

closed 2016-10-16 by yasu80