CIR-KIT/fifth_robot_pkg #4: ブランチ修復中

HajimeTokuyasu, 2016-09-15:
ファイル整理中に要るファイルまで強制プッシュで消滅させてしまったので,しばらくpullとか待ってください....

forno, 2016-09-15:
こっちの手元のリポジトリをforce pushしますか？
バックアップは二重にありますので。

ちなみにrevertとrebaseは全くの別物ですのでご注意ください。

HajimeTokuyasu, 2016-09-15:
お願いします.

forno, 2016-09-15:
force pushを正常に行えたことを確認しました。

Settingにbranchesの欄があり、Force pushに対して Protectをかけれるので、やっておくといいかもしれませんね。

forno, 2016-09-15:
おそらく今回のてんやわんやは
git pushしようとしてConflictしたのでgit pullしようとして更に面倒なことになったのかと思われます。

うちらの推奨設定だとpushがmerge, pullがrebaseになってるので、push(merge)にConflictしたからといってpull(rebase)しようとするとより発狂死するような地獄に会います。

もともとブランチを切って作業するのが一番ですが、
mergeのConflictのほうが簡単ですので手動でremoteのbrancheにmergeしてからpushするのが一番安全です。　場合によっては履歴が汚くなりますが。

長文失礼しました。

forno, 2016-09-15:
伝え忘れていましたが、これでIssueの回答(必ずしも"解決"に限らないが大抵はそのこと)が出たらIssueを閉じましょう。

この時コンセンサスを取るとなお良しですが、基本的にIssueの責任者が回答が出たと思った時に閉じて良いと思います。

HajimeTokuyasu, 2016-09-15:
訂正します.

closed 2016-09-15 by HajimeTokuyasu