CIR-KIT/fifth_robot_pkg #6: 開発メンバーの中に Kinetic Kame を未だインストールしていない人がいれば問題です

HajimeTokuyasu, 2016-09-15:


forno, 2016-09-15:
電装の概要図の作成をおなしゃす。

イメージとしてはこんな感じかなと。 https://gist.github.com/forno/6a158820c1ff691f43a25ec64d277674

forno, 2016-09-18:
Notice:
https://github.com/CIR-KIT/fifth_robot_pkg/commit/7f6994d1ceeca4ce546dc610891ced4831e916d1
上コミットで電源取り回しが示されました。

電子回路はまた別にアップロードしてください。　その時はなるべく一般的な回路図を用いてください。

yasu80, 2016-09-18:
わかりました. 
#CADるなら標準ソフト決めません?

yasu80, 2016-09-18:
電装系の配線については, 大まかには
 ・ 誰かのUbuntu-PCがLRFとボードに接続,
 ・ ひょっとしたら, 本当にひょっとしたらWEBカメラが乗るかもです

forno, 2016-09-18:
ここは電装用IssueなのでCADは使わないかな。
一応答えると、共通の拡張子が扱えればどれ使ってもいいとおもうので、ソフトを制限する必要はないかと。

forno, 2016-09-20:
https://github.com/CIR-KIT/fifth_robot_pkg/pull/8
電装修正しました。

forno, 2016-09-23:
特にfig_power_supply_connection以外に回路が必要で無いならこのIssueは閉じて構いません
制御はまた別のIsuueを開きましょう

yasu80, 2016-09-23:
いえ, おそらくバッテリ回りを弄くりますので開けておきます.
具体的には, あの機体の標準状態はバッテリへのアクセス性が最悪です.
つまり, 前輪を取り外すか, 非常に微妙な釣り合いを保つバーをネジで外すかしないと取り出せません

上面にトラップドアを設けて, 回路を切り開き, 更にそこからスイッチを伸ばそうと考えています.
その辺の実装図は後にあげます.

forno, 2016-09-24:
https://github.com/CIR-KIT/fifth_robot_pkg/projects/1#card-233693
電装の満たすべき要件のCardはこちら。

Projectの方が更新されても広告されないので、こうやって引き合いに出すか、
Issueとして適当なら(例えば電装のチェック要素の確認とか)Issueを作ってProjectに登録する方がよいかと。

AriYu, 2016-10-04:
すみません。ちょっとこっちでの議論に気づきませんでした。
ちょっと電装の図を見て思ったことを書きます。
緊急停止スイッチが大本の電源の直後に来ていますが、この構成であれば緊急停止を押した瞬間にセンサ類も落ちてしまいます。
つくばチャレンジでは緊急停止ボタン押下後の復帰が認められることがあります。その際、緊急停止スイッチを戻すだけで復帰できることが理想だと思います。
そのため、緊急停止スイッチを押してもセンサ類が死なないような構成にすることが望ましいと思います。
あとLRFに渦電流が流れると壊れてしまいますので、何でもいいですがその辺の対策もお願いします。

yasu80, 2016-10-04:
> $B$9$_$^$;$s!#$A$g$C$H$3$C$A$G$N5DO@$K5$$E$-$^$;$s$G$7$?!#(B
> $B$A$g$C$HEEAu$N?^$r8+$F;W$C$?$3$H$r=q$-$^$9!#(B
> $B6[5^Dd;_%9%$%C%A$,BgK\$NEE8;$ND>8e$KMh$F$$$^$9$,!"$3$N9=@.$G$"$l$P6[5^Dd;_$r2!$7$?=V4V$K%;%s%5N`$bMn$A$F$7$^$$$^$9!#(B
> $B$D$/$P%A%c%l%s%8$G$O6[5^Dd;_%\%?%s2!2<8e$NI|5"$,G'$a$i$l$k$3$H$,$"$j$^$9!#$=$N:]!"6[5^Dd;_%9%$%C%A$rLa$9$@$1$GI|5"$G$-$k$3$H$,M}A[$@$H;W$$$^$9!#(B
> $B$=$N$?$a!"6[5^Dd;_%9%$%C%A$r2!$7$F$b%;%s%5N`$,;`$J$J$$$h$&$J9=@.$K$9$k$3$H$,K>$^$7$$$H;W$$$^$9!#(B
> $B$"$H(BLRF$B$K12EEN.$,N.$l$k$H2u$l$F$7$^$$$^$9$N$G!"2?$G$b$$$$$G$9$,$=$NJU$NBP:v$b$*4j$$$7$^$9!#(B
>
> -

$B5.=E$J%"%I%P%$%9$r$"$j$,$H$&$4$6$$$^$9(B.
$BBg2q%k%k%V$N:FEY$N=OFI$H:GE,2=$O=EMW$G$9(B.$BEE8;$K$D$$$F$O$=$N$h$&$K(B, $B$[$+$b:FDj5A$NI,MW$,$"$k$+3NG'$7$F$*$-$^$9(B.

Outlook for Android<https://aka.ms/ghei36> $B$+$i<hF@(B

forno, 2016-10-04:
https://github.com/CIR-KIT/fifth_robot_pkg/pull/16

Issueを受け取ったので、現在のヒューズ状態の明記と緊急スイッチの移動を行ってみました。

forno, 2016-10-04:
私にはLRFの耐久電流値と必要電流値がわからないので、そもそも1Aで足りるのか、あるいは大きすぎるのかは存じ上げませんが、図に示した通り12V側のFuseは1Aです。 @yasu80 問題が無いかチェックしてください。

yasu80, 2016-10-05:
確認しました. 問題ないと思います.
[URG公式](http://www.hokuyo-aut.co.jp/search/single.php?serial=146)では 700mA 駆動 / 1A で起動のよう指示が出ています, 最大 1A と見ていいはずです.

LRFへの電源供給は電源基板側[公式](http://t-frog.com/products/power_supply/)の定格出力が 12V / 1A なので特に心配はしていません.

yasu80, 2016-10-05:
確認しました, 問題ないです.
電源の[定格](http://t-frog.com/products/power_supply/)が 12V /  1A あるのに対して
URGの[定格](http://www.hokuyo-aut.co.jp/search/single.php?serial=146)が 12V / 700mA ~ 1A ということで
このまま運用することができます.

yasu80, 2016-10-08:
一段落したので閉じます.
具体的なスイッチの改造とかについては別途issueを立ててください.

closed 2016-10-08 by yasu80