CIR-KIT/fifth_robot_pkg #9: LICENSE 選択肢

forno, 2016-09-19:
1. No licensed
2. Licensed same license
3. Unlicensed

Licenseは上の3つに分けられるかと思います。

1のライセンス無はライセンスはあげないと明記するとなおよしです。
ただしGithubの規約で無断でForkと閲覧はできます。現在の状態です。

2は各ライセンス下で利用OKと示します。ライセンスごとに特徴が異なります。

3はPublic domainと言って全著作権を手放します。無保証であることを宣言する文章は含めます。

好きなのを選んで一応みんな書いておくといいのではないかと思ってIssueを開いてみます。

yasu80, 2016-09-19:
重要なのは管理コストではないかと思います.

権利侵害について実際そこまで神経質になることもないでしょうし, ライセンスつけるにせよ規約の少ないや奴 (メールが来ない奴→管理が楽) にしておきたいです....MITとかBSD二条項とか

forno, 2016-09-20:
まあ、どちらかというと我々が権利侵害をしないように気を付けようぜ！
-> 自分でライセンスつけたら勉強になるだろ。引用元のライセンスも見るし。
というところはある。

あとGit moduleで引っ張ってきてる先のライセンスを表示しないといけないのかどうか気になる。

あとせっかくGithub使ってるのでいろんな人の勉強になるようにって意味合いも込めて。

forno, 2016-09-20:
管理コストについては(一部の2を除いて)どれを選んでも同じでしょう。
ソースコードの先頭に全部ライセンス表示つけると面倒ですが、大抵は(自分の物には)つけなくてもいいみたいですし。

基本的に、LICENSE.txtとREADMEへの明記だけですよー。

forno, 2016-09-20:
個人的にはROSがBSDですし、catkin系列のソフトは全部BSD Licenseでリリースするのがいいと思ってたり。

参考:
http://docs.ros.org/diamondback/api/licenses.html

forno, 2016-09-23:
とりあえず BSD-2-Clause License でも張っておきます？

yasu80, 2016-09-23:
そうしましょう. 張ってきます

forno, 2016-10-01:
ぷるリクが通ったので閉じます。

closed 2016-10-01 by forno