CIR-KIT/fifth_robot_pkg #11: 頭でっかちすぎる

yasu80, 2016-09-23:
こいつを見てくれ・・・・

[image]

 ・ パソコンがでかい
 ・ 足回りがおそらく重量 (3kg+バッテリ+固定具) でヘタってしまう
 ・ パソコンの強度に不安がある, こいつは以前, バイト先の馬鹿がひったくろうとしてヒンジが割れた経緯がある
 ・ よしんばアブソーバ, 固定具, その他もろもろを装備したとしてもバランスが悪い (トルクの発生部に対して重心が高いと, いわゆる不安定な釣り合い状態になってしまう)
 ・ パソコンがでかすぎる
 ・ 処理力と安定性にも疑問がある. よくハングるので本番環境で私の Vostro くんは持ち出したくない

→一年生のハイスペックなやつを誘拐してくるのが私の希望

forno, 2016-09-23:
そうですね。まずいえるのはセンサが固定されていないのでセンサの調節ができなさそうってことですかね。

足回りがどれだけ持つかわかりませんが。そこら辺は概算でいいので計算してみてください。
やり方は私にはわかりかねるので必要であれば @shimoe や @Dsaikoro 、あるいは先輩に聞いてみてください。

環境は性能の高く、かつめっちゃやたら軽い一回生のパソコンがベターだと思います。

まずは、必ず定位置に置かなければならないものを固定する方法を考えるべきかと思います。

yasu80, 2016-10-13:
[image]

バッテリがバッチリ付いたと言いたいところですがいくつか課題があり,

1. 現状仮止めであるから強度が不足である
1. 現状固定位置が低すぎて腹が擦りそう
1.  現状固定位置が後ろ過ぎて従輪が当たりそう
1.  現状片手支持であるから妙な曲げモーメントがかかってよくない
1.  バッテリがのっかってるだけ

しかるに,

1. 増し締めと必要に応じワッシャ/スプリングワッシャの追加
1. スライダ部分(高さを稼いでいる部分)をずらして上下調整
1. フレーム側のスライダをずらして前後調整
1. もう一個後方につけてあるアングルから支持架を伸ばして両手保持とする
1. 効率の良い緊縛方法の考察

を一回生ハード班に要求したいと思います.  もちろん大急ぎで.

yasu80, 2016-10-13:
あとアングルの固定力は絶大でしたから, LRF( LaserRangeFinder ,レーザー測距儀) の固定もこれを活かせば上下調整容易く把持確実なること間違いなかろうと思います. これも急ぎで.

yasu80, 2016-10-16:
大方の構造は河野氏含む一回生の協力で完成しました. ということで作業に区切りをつけてここで一旦 Closed とします.

closed 2016-10-16 by yasu80