CIR-KIT/fifth_robot_pkg #14: new_pkgs_drafting

yasu80, 2016-10-02:
諸事情あってソフトを既存品使用から新造に舵取りすることにしました.
その要件定義・実装までをこのissueでメモしていきます.

yasu80, 2016-10-02:
###使用するべき既存レポ
 ・ hokuyo_node(LRF)
 ・ navigation(ウェイポイント管理)
 ・ gmapping(地図作成)
 ・ yp-spur(i-cart mini プロジェクトの方で提供されるt-frogドライバ制御・オドメトリ取得用の製品)

現状, GPSモジュールのたぐいの入手あてが無いので, 今のところはGPS活用はソフト面では予定しながら実装面では見送ることになるでしょう.

暇があればwebカムとOpenCVライブラリを投入して人探しの課題もやってみたいのですが, 現状めどが立っていない悲しさ. 夢のまた夢というのが本当のところ.

###運用
 ・ 試走日に人間がついていってコントローラで走らせつつ, LRFでデータを取得する - 「犬の散歩」 をやることでgmappingから地図データを作らせる
 ・ 自己位置推定をオドメトリによって行いつつ, LRFで障害物回避を実行, 走行.

forno, 2016-10-09:
現在のpackage構成

Thard patry softwares (other made)
- navigation
- hokuyo_node
 + driver_common
- ypspur_ros_bridge

We make their
- goal sender #20
- configuration (for navigation)

navigationを中心とした構成で挑みます。
[navigationのwiki](http://wiki.ros.org/navigation/Tutorials/RobotSetup)
をよく読み、必要な設定、パッケージ、メッセージ変換を行う必要があります。

それぞれ、目標位置の設定だったり、障害物があるときの回避の仕方だったりと、
作る箇所がどこに対応しているか確認しながら作業を進めることが大切です。

これらソフトウェアについては @forno (Doi Yusuke)や @yasu80 (Hajime Tokuyasu)に聞けば開発に参加できます。

forno, 2016-10-09:
# configuration (for navigation)の詳細

- Transform Configuration [**We make this**]
- Sensor Information -> hokuyo_node
- Odometry Information -> ypspur_ros_bridge
- Base Controller -> ypspur_ros_bridge
- Mapping (no required ?)

-> が存在するところは右側のパッケージが行ってくれるという意味です。
Mapは必要と書いてあったり書いてなかったりするので後ほど詳しく調べます。

御覧のようにtfのデータを作成する必要があります。
これは機体の構成と、センサの値について行う必要があります。

参照先は同様に[navigationのwiki](http://wiki.ros.org/navigation/Tutorials/RobotSetup)です。

forno, 2016-10-15:
packageの構成は固まったので、fraftingの段階は抜けたと判断します。
このIssueは閉じてはいかがでしょうか？

yasu80, 2016-10-16:
そうですね, そうしましょう.
以後は issue と各作業について一対一対応で細かに取り上げていこうかと思います.

closed 2016-10-16 by yasu80