CIR-KIT/fifth_robot_pkg #15: readmeにおける図の間違い？について

AriYu, 2016-10-04:
issue立てるほどのことでもないですが、少し気になったので指摘しておきます。
topのREADMEにある図にある地図に関してですが、
http://wiki.ros.org/navigation/Tutorials/RobotSetup?action=AttachFile&do=get&target=overview_tf_small.png
にあるように、自律移動中にnavigation stackに地図(占有格子地図)を提供するノードは`map_server map_server`ノードです。
一方、`map_server map_saver`は地図作成中(gmapping中）に実行することで地図を保存するノードです。
gmappingで地図を作成中に`rosrun map_server map_saver -f filename`とすることで地図を保存することができます。
地図は一対のyamlファイルとpgmファイルからなります。`map_server map_server`が読み込むのはyamlファイルですが、地図の実態はpgmファイルです。実際には単なる画像ファイルなので画像処理ソフトで編集することも可能です。yamlファイルの中身も確認して何が書かれているのかを確認しておくと良いと思います。

できれば修正の方をお願いします。

yasu80, 2016-10-05:
わかりました.

あと, kineticにurg_nodeが対応していないようなので再度書きなおしておきます.

yasu80, 2016-10-05:
修正しました. 
多忙な中, 貴重な指摘に感謝します.

closed 2016-10-05 by yasu80