CIR-KIT/fifth_robot_pkg #17: 新しいbranch運用について

yasu80, 2016-10-05:
いまのところ develop ブランチには当初の計画に従った, 調整中の作業が残っていますが, 
これとは別に, new_pkg ブランチに新規作成中の作業が存在します.

古い方を残したほうがいいかもしれませんので develop を old-develop として新規にログとして残し,
new_pkg ブランチを develop ブランチへと統合して上書きすることを提案します.

forno, 2016-10-05:
残す必要性を感じないでござるよ。普通にdevelopにマージして続き書いてけばいいと思うです。

forno, 2016-10-05:
今気づいたけど相当gitmoduleが誤ってるのでfixしときますね。

yasu80, 2016-10-05:
どっちのbranchを変更中ですか? mergeの途中で気がついたんですが, 激しくconflictしてませんか.

forno, 2016-10-05:
なんでconflictしてるのかと思ったら両方Readme触ってるせいじゃねぇかｗ　特にURLとかかな？

yasu80, 2016-10-05:
あーはいそうです, 画像リンクのところがファイル名変えてるんで

yasu80, 2016-10-05:
いまさらですけれど, common_driver はLRFを kineticで稼働するために必要なドライバ類のパッケージで, 標準で入ってないので取り込んでおきましたが, この中に含める必要なかったですね

forno, 2016-10-05:
というかなんでトップに置いたし、ってだけ。srcの中にもいたし。

forno, 2016-10-05:
pull-requestで修正するのが面倒だったのでmerge-newpkgブランチをきってdevelopマージ版作りました。

forno, 2016-10-05:
なお、気づいたと思いますが、image専用にブランチ作りました。

forno, 2016-10-05:
https://github.com/CIR-KIT/fifth_robot_pkg/pull/19
こちらのpull-requestはコンフリクトせずに引っ張り込めますー

yasu80, 2016-10-05:
受けました.
死んだかと思った...

closed 2016-10-05 by yasu80

forno, 2016-10-05:
conflictしたらconflictしたファイル(both modify)を編集してgit add してcommitするだけなので今度やるといいよ
あと何も編集せずにaddしてcommitしてその後のコミットでconflict部分を直すのも手かと。

yasu80, 2016-10-05:
いまログよんでます. 後はmerge作業開始時点でissueにかくとかですね. ありがとうございます.