CIR-KIT/fifth_robot_pkg #20: goal設定機の設計

yasu80, 2016-10-08:
goal設定機仕様書:
waypointを(yaml/csv/...)などで記述
→ 手動 
→ 記述ソフトつき (rviz とかで組むか?)
waypoint更新条件の判定
← odomを読む
← amclからtfメッセージを読む
waypointを随時更新

forno, 2016-10-09:
用語goalとwaypointを5号機では区別します。

とりあえずの解釈として

* waypointはコースを指示するもの
* goalはnavigationに渡す目標位置

とします。

yasu80, 2016-10-11:
yaml 読むところは python で実装すればOK. 連想配列なんかの機能がいい感じに生きる.
コース計算に必要な線形性のチェックあたりは今日中に頭をひねるとする.

yasu80, 2016-10-11:
yaml の形は確定. 素晴らしく扱いやすいと思う.
python からの読み方もわかったから, あとはどこまで python にやらせるか,のみ.
実際 あとはトピックでも何でも他のモジュールに計算させてもいいし.

//以下愚痴
さっと行けそうだから python 実装やるっていったけど, 思いの外 rospy の設定とかPyYAMLの動き方とかあれやこれや手間取るっすね...これふつーにC++でもCでもデコーダ自分で書いたほうが楽かも分からない

けれどもここまでやったからせめてデコーダくらいは python で作ろうと思う, 辞書型勉強するの面白かったし...(典型的なコンコルド効果).

forno, 2016-10-15:
param server から受けとった複雑な連想配列をc++で扱うのは死ぬほど面倒です。
あらゆる連想配列を扱えるオブジェクトを構造が間違っていないか石橋を叩きつつ作業する必要があります。
パラメータが2段以上入れ子になる場合はpythonを使うことをおすすめします。
逆にパラメータが連想配列一段分 ex) map<string, double> である場合はc++で書くほうが優しいかと思います。

forno, 2016-10-29:
@yasu80 
Waypointとして読み込むファイルの形は確定しています。
すなわち、これは読み込むファイルのメッセージ型を定義できることを意味しています。
また、その場合そのメッセージの配列を作ることは非常に容易です。

ということで考えた構成が次になります。

yaml_readerさんはyamlファイルを読んで(あるいはrosparamから読み取って)そのメッセージ型に変換し、トピックあるいはサービスで外部に公開します。
これであれば処理をする部分をc++にすることもできれば、pythonにすることもでき、とても嬉しいかと思います。

forno, 2016-10-30:
オドメトリがある程度適当で良い理由。
1. 走行中に-360°〜360°以上回転する可能性が少ない
2. 同じコースを走って同じオドメトリが取れるならロボットは(ロボットにとっては正常な)歪んだオドメトリ上の位置を把握できる。
3. 上と関連して、多少歪んでいてもロボットにとっては正常な自己位置推定機能が働く。

と言った要因がありそうな気がする。(気がするだけ)

forno, 2016-11-13:
とりあえずゴール設定機(goal_sender)はかんせいしましたので閉じます。
yaml_readerがんばってくださいまし。

・・・とはいえ、yamlリーダーのもっとスマートな案を思いついてしまいましたが。

closed 2016-11-13 by forno