CIR-KIT/fifth_robot_pkg #23: 高さを稼ぐ形状を確定してください. note: アルミ材はジャンクヤードにあります.

yasu80, 2016-10-15:


rkdora, 2016-10-17:
ナフコにて
ロボットの外枠にポリバケツとかちょうどよさげでした
[image]
[image]
[image]
[image]

[image]
[image]
[image]

yasu80, 2016-10-17:
視察ご苦労様です.
いろいろ問題点がありまして,

 - なるだけ工場にあるものを活用しませんか・・・
 - フレームと一体ではない, しかも強度的に信頼のならないゴミ箱めいたプラッチックをかぶせたとして緊急スイッチ押下の衝撃力はどうなるのか? (これは暴走時に押すスイッチで, 暴れるロボットを殺すためのものであることをお忘れなく)
 - 大きなカバーをかぶせたとしてメンテナンス性をどう確保するのか? 着脱式にするとして固定どうするのか?
 - 技術力の不足を感じさせる

などが考えられます.

このロボットはつくばで実環境運用ロボットの試験に供するための機材です. 故にフレーム一つにもそれ相応の最適化が望ましいことは言うまでもありませんし, ありあわせの部品を使うのはともかく, 安易に最適でない構成をなすことは避けてください.

再考を要求します. フレームの組み方なら教えられますし, 必要な部品はかき集めましょう...

yasu80, 2016-10-29:
@rkdora 君, @AsukaYamasaki 君, @tanacchi 君他多数の活躍でいい感じになりました.
ありがとう.
[image]

closed 2016-10-29 by yasu80