CIR-KIT/fifth_robot_pkg #27: 走行テストと map データ取得について

yasu80, 2016-10-23:
やっておくと勉強になりそうなことなんですが,
走行試験とマップの取得をやる必要があります.

## 手順

openEth.sh を実行
mapping.launch を実行ののち,
rosbag record -O mapdata.bag /scan /tf

この状態で走行し, 走行 / rosbag を終了の後

roscore
rosparam set use_sim_time true
rosrun gmapping slam_gmapping scan:=scan
別ターミナルで
rosbag play --clock  mapdata.bag
終了の後, **閉じずに**
rosrun map_server map_saver -f map_factory

でマップが出てきます.
※ -f map_factory の部分は出力ファイル名指定. 変えてね

## 現状

現状, map は取れてるけれど非常に汚くて使い物になりません.
urg_node 側のパラメータもいじって良いデータを取得できるようにいろいろ試しつつ調整する必要があります.
具体的には, パラメータファイルの調整とか,
読み取るトピックを/scan にするのかマルチエコーモードを有効にして /last にするのか  /first にするのか /most_intense にするのかって話だったりもします.

自立走行の根幹をなす部分です. ぜひ参加しましょう.
私はほぼ毎日いるはずなので, 来てくれればやることは見つかるんでよろしく.

forno, 2016-10-29:
オドメトリのキャリブレーションの精度が高まってきたため、マップの綺麗さが上がってきました。
運用できる最低限ギリギリの精度であると思います。しかし、まだまだ精度を上げたいと思います。

今日(日曜日)に行いたいことをまとめます。

1. 最新マップでのnavigation挙動確認(rvizで指示しての動きが正常であるか)
2. 屋外でのオドメトリ精度の確認
3. オドメトリ精度向上(パラメタファイルの最適化)

以上の３点は梱包され触れなくなると想像と推測でソフトを開発しなければならなくなり、大変困ります。
@tanacchi には悪いけれど、できうる限りよろしくお願いします。

AriYu, 2016-10-29:
できれば現状の地図をみたいのですがどこかに上がってたりしませんか？
あとbagファイルもください。

forno, 2016-10-29:
@AriYu 連絡ありがとうございます。
## bagファイルについて
bagファイルはあまりにも巨大であり、同一リポジトリに込めると`clone` や `pull` に甚大な影響が出るため現状はアップロードしておりません。また、特に今回マップをとった最新のbagファイルはすでに破棄されているため渡すことが不可能です。

## mapファイルについて
現状の最新版は以下のディレクトリに入っています。 mapだけとなりますがどうぞご覧ください。
https://github.com/CIR-KIT/fifth_robot_pkg/tree/develop/maps/factory/tanacchi3

AriYu, 2016-10-29:
@forno 
ありがとうございます。
bagファイルの件、分かりました。
次取ったらgoogle driveか何かにアップロードしてください。
よろしくお願いします。

forno, 2016-10-29:
@AriYu
わかりました。そのように通達します。

forno, 2016-11-01:
依存関係がごちゃごちゃしてた上にその解決が面倒であったため、goal_senderをできうる限り依存を小さくして作りなおしました。
本当はCSVに相当するデータも別から流し込みたいですね。
yaml_readerさんと連携すれば結構幸せになれるんじゃないかと思ってみたりしています。
今後の課題としておきませう。

closed 2016-11-13 by yasu80