CIR-KIT/fifth_robot_pkg #28: 調整後のパラメータを使うと Buffer Overflow が発生している

yasu80, 2016-10-29:
先日 @tanacchi 君と作ったパラメータファイルを私の環境で 実行したところ, なんか怪しいエラーが出ました.
以下のような具合です.
マジで意味不明なんですけれど, とにかく 前回のコミットへ checkout してごちゃごちゃやってます.
向こうの環境をもう一度確認してみたい感じですが・・・

yasu16@yasu16-Vostro-3546:~/Documents/workspace/robots/cirkit5/fifth_robot_pkg/src/fifth_robot_launcher$ roslaunch mapping.launch 
... logging to /home/yasu16/.ros/log/ee9b6f0c-9d7e-11e6-a961-74e6e24de61a/roslaunch-yasu16-Vostro-3546-2978.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yasu16-Vostro-3546:35341/

SUMMARY
========

PARAMETERS
 * /joy_node/autorepeat_rate: 20
 * /joy_node/deadzone: 0.3
 * /joy_node/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.5
 * /teleop_twist_joy/axis_angular: 0
 * /teleop_twist_joy/axis_linear: 1
 * /teleop_twist_joy/enable_button: 8
 * /teleop_twist_joy/enable_turbo_button: 10
 * /teleop_twist_joy/scale_angular: 0.4
 * /teleop_twist_joy/scale_linear: 0.5
 * /teleop_twist_joy/scale_linear_turbo: 0.5
 * /urg/angle_max: 1.5
 * /urg/angle_min: -1.5
 * /urg/ip_address: 192.168.0.10

NODES
  /
    joy_node (joy/joy_node)
    laser_transform (tf/static_transform_publisher)
    teleop_master_node (teleop_master/teleop_master_node)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
    urg (urg_node/urg_node)
    ypspur_coordinator (ypspur_ros_bridge/ypspur-coordinator.sh)
    ypspur_ros_bridge (ypspur_ros_bridge/wait_and_ypspur_ros_bridge.sh)

auto-starting new master
process[master]: started with pid [2991]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ee9b6f0c-9d7e-11e6-a961-74e6e24de61a
process[rosout-1]: started with pid [3004]
started core service [/rosout]
process[ypspur_coordinator-2]: started with pid [3007]
*** buffer overflow detected ***: ypspur-coordinator terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f845d2937e5]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7f845d33456c]
/lib/x86_64-linux-gnu/libc.so.6(+0x116570)[0x7f845d332570]
/lib/x86_64-linux-gnu/libc.so.6(+0x1158c2)[0x7f845d3318c2]
ypspur-coordinator[0x405726]
ypspur-coordinator[0x402712]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f845d23c830]
ypspur-coordinator[0x4030e9]
======= Memory map: ========
00400000-00415000 r-xp 00000000 08:05 2232790                            /usr/local/bin/ypspur-coordinator
00614000-00615000 r--p 00014000 08:05 2232790                            /usr/local/bin/ypspur-coordinator
00615000-00616000 rw-p 00015000 08:05 2232790                            /usr/local/bin/ypspur-coordinator
00616000-00617000 rw-p 00000000 00:00 0 
00b62000-00b83000 rw-p 00000000 00:00 0                                  [heap]
7f845d006000-7f845d01c000 r-xp 00000000 08:05 2626581                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f845d01c000-7f845d21b000 ---p 00016000 08:05 2626581                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f845d21b000-7f845d21c000 rw-p 00015000 08:05 2626581                    /lib/x86_64-linux-gnu/libgcc_s.so.1
[... 53 lines cut ...]
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
cmd: 0.000000, 0.000000
vel: -64544568146914994148182327296.000000 [m/s], -0.000000 [deg/s], odom: 0.000000 [m], 0.000000 [m], -nan [deg]
^C[teleop_twist_joy-8] killing on exit
[joy_node-7] killing on exit
[teleop_master_node-6] killing on exit
[laser_transform-5] killing on exit
[urg-4] killing on exit
[ypspur_ros_bridge-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
yasu16@yasu16-Vostro-3546:~/Documents/workspace/robots/cirkit5/fifth_robot_pkg/src/fifth_robot_launcher$

yasu80, 2016-10-29:
ちなみに 前回コミット分までの環境に戻せば, ypspur-coodinator ros_bridge teleop 以下全部まともに機能していますから, 多分変更分になんかあるんだと思います.

odometory とってきた部分の deg が　NaN返してるんが割と気になりますが

yasu80, 2016-10-29:
作業メモ:

デフォルトファイル/旧 mapping.launch -> ok
デフォルトファイル/新 mapping.launch -> ok
改変ファイル/旧 mapping.launch -> 死亡

多分改変ファイルの内容か, 環境のどこかに不整合があるのかもかも

forno, 2016-10-29:
対応：Parameterファイルを破棄して再度調節開始

forno, 2016-10-29:
解決:新ファイルのキャリブレーションが成功したため件のパラメータファイルは永久破棄

closed 2016-10-29 by forno