CIR-KIT/fifth_robot_pkg #35: Pull-requestの受け取り方

forno, 2016-11-02:
なんか適当にマージしとけみたいな感じが出たり出なかったりします。
現状を改善すべく、Pull-requestを有効活用するための方法を考えましたので少し聞いてくださいな。

Pull-requestにはレビュをするための機能がついています。
また、Pull-requestが正常で、マージしてもいいと判断したことを明示する方法もあります。

そこで、まずはマージ許可(approve)を活用してゆきましょう。
他人のコードを読む良い機会にもなりますし、逆に、コードを読まれる良い機会にもなります。

そのために、次の制限を提案します。
1. Pull-requestのマージはPull-requestを出した人が行う
2. Pull-requestは誰かにマージ許可(approve)されてから行う

この2点です。別に今年これからしろとは言わないので、気に留めて置いていたただけましたら幸いです。

forno, 2016-11-02:
あとPull-requestの返信でなく、マージコミットの本文に返信を書くのをやめてください

yasu80, 2016-11-02:
上については,
 1, Pull-req の使い方
は　#22 で言っていた機能のことでしょうか? あれって保護ブランチへの取り扱いですよね?
 2, approve の使い方
は,  割と誤操作防止用程度にしか考えてなかったのでそのように活用できれないいと思います.

下については......連続誤爆です...謝罪

forno, 2016-11-02:
別に保護ブランチに限らず、他者の目に一度晒されてからのほうが何かと都合が良いのです。
でないとよくわからない機能が勝手に追加されてよくわからないうちに動きそうなので。

1.のPull-requestを出した人がマージするのは、マージした後に問題が出てないか一番正確に確認できるのがPull-request出した人だからだろうということがひとつ目の理由ですかね。(もちろんこれは手元で一回お試しマージしてから正式にPull-requestを受けるのが正解だと思います)

ふたつ目の理由としては、開発中にレビュだけ受けたい時があって、その時に気軽にPull-requestを投げたいからです。
誰かが困っているところをみんなと共有できて、どんなところで躓いたかログが残るうえに解決の糸口が広がる可能性が高まります。

そのための2つのPull-requestお作法を浸透させて行ければ良いかなという考えであります。

forno, 2016-11-02:
あー。返信し忘れ。

> 1, Pull-req の使い方
は　#22 で言っていた機能のことでしょうか? あれって保護ブランチへの取り扱いですよね?

#22 で言った機能はマージにAppoveを必須とするものです。
ただ、レビュしてくれる人がいなさそうな気がしたので、イライラしないようにdevelopではオフにしてあります。
ただ、まあ、レビュしてくれる人がいたほうが嬉しいなって。
レビュの内容は、「こっちの環境でも動きました」だけでも死ぬほど助かるので。いやマジで。

yasu80, 2016-11-03:
1, approveボタンが見当たらないってのと,
2, これ見てると, 
[参考](http://qiita.com/terra_yucco/items/fa08bd2a4b498963a313)
approve 機能は保護ブランチで有効な気がします

forno, 2016-11-03:
保護のレベルは自分で選べるのです。
今はmasterにforce pushとapprove無しマージ禁止、developにforce push禁止をかけています。
レビュの仕方は今からスクショとって説明します。

forno, 2016-11-03:
はじめにプルリクエストの初期画面があります。
[image]

変更の確認に移動します。File changedってところをクリックしてください。
[image]

右上のレビュボタンを押して全体に対するレビュを行います。
[image]

また、変更コード内の任意の行についてもクリックすることでその行に対するレビュを行うことができます。(このｔきは変更の要請やコメントだけでApproveはできない)

forno, 2016-11-13:
これから先Github flowにに切り替えた後はレビュを必須に設定しておきますー。

closed 2016-11-13 by forno