CIR-KIT/fifth_robot_pkg #37: goal_senderの使い方が分からないです

tanacchi, 2016-11-03:
マップを取得したあと、マップを読んで自立移動させる方法がわかりません。 rosrun を試してみましたがうまく行きません。徳安先輩にも相談してみましたが、解析できないそうです。 launch ファイルが間違っていると思われるそうですが、、、、

yasu80, 2016-11-03:
・ map_server を起動しておく
 ・ third_party にある waypoint_server を立ち上げる
 ・ 例の新作を立ち上げる

で良かったと思う. 頭がシャンとしてないのでまちがってるかも.
launch は そのうち上げる

forno, 2016-11-03:
おっと。ドキュメントまとめてなくてすみません。

ノードのローカル名前空間のpathパラメータにcsvファイルを投げ込めば動くとおもいます。
ぶっちゃけ動作自体は元にした奴と同じはずなのでそちらのドキュメントも役立つと思います。

forno, 2016-11-03:
自立移動に使ってるlanchファイルってnavigation.launchですか？
見たところwaypointServe.launchがろくに記述されてないようなのでこれを修正します。

あとそれに伴ってwaypointファイルをgoal_senderパッケージに追加します。

forno, 2016-11-03:
#40 にて動作すると思われる状態のlaunchファイルを追加しました。

forno, 2016-11-03:
#40 にて navigation.launch 一個で自動運転が開始されるように修正し(たつもりになってみ)ました

tanacchi, 2016-11-03:
夜遅くまで対応ありがとうございました。。。

forno, 2016-11-03:
はーい。　残りのpull-requestも受け取ってくれたらWaypointのメッセージがより見やすくなるので適当にマージしていいですよー。

forno, 2016-11-04:
goal_senderの使い方にて、未だに不明な点はありますか？

tanacchi, 2016-11-04:
外でまともな地図が作れてないのでまだその段階にいってません。。
もうしばらくしたら宿舎のロビーを使ってｇｏａｌ_senderについての確認もろもろをしますので
不明点についての報告はもう少しあとになりそうです
いまから今日とれたｍａｐをaddしようかと思います

forno, 2016-11-04:
さんくすです。こっちでも確認してみます。

forno, 2016-11-04:
そういえば、恐らくですがFixのプルリクを受けてない場合はゴールして停止した直後にセグフォで落ちるかと思います。

forno, 2016-11-05:
いかがお過ごしでしょうか

yasu80, 2016-11-05:
自律走行まで成功しました.

waypoint 記録ツールにより コースを記録した後に, navigation.launch により自律走行を行いました.
良く地形, コースに追躡しましたが, map 内に残っていた未探査領域に突入してスタックしてしまうため十メートル前後の地点で繰り返し死にました.

 ・ 自己位置推定の様子がおかしい. amcl がコース外へアウトすることが多発するし, ひどい時は引き返したりする
→　わかりやすいランドマーク付近のみを走行するようにし, 走行パターンは一定かつ安定したものをサンプルとすることできちんとしたコースcsvを取得できた
 ・ LRF がまだ地面を読んでる. 空を見せてもしょうがないし, かと言って地面を見せてもいけない.あ実際, この測域センサは水平にビームを飛ばせることを前提としているので振動対策を講じなければならない
→走行の方式を上と同様に工夫したら割となんとかなった.

forno, 2016-11-05:
おおー。おめっとさんなのです。　reopenは何か思ってのことです？

yasu80, 2016-11-05:
誤爆です(小声

forno, 2016-11-05:
Closeのほうが誤爆なんですねー。
とはいえgoal_senderの使用法的にはもういいかと思うので、報告用Issueを作りませう。
作るならそちらでおねがいしますー

forno, 2016-11-05:
にゅーん。まだ何か問題ありますー？

forno, 2016-11-13:
問題なくなったらきちんと閉じてくだち

closed 2016-11-13 by forno