CIR-KIT/fifth_robot_pkg #43: [募集]五号機の問題点全般

yasu80, 2016-11-04:
* 外装について, 『フレーム,機能部とハリボテ』思想では不十分である. 強度, 耐力, 保護能力, 維持管理その他全てについて不十分である.
	- 専用設計で作り起こす, 現状 FRP で専用に固めて作るのが有効かと思っている.(趣味用に一般的で, 完全な防水性を持つ)
	- 点検ハッチとかが付けたい
	- 役物をすべて正しい位置に格納できるようにしたい. そして全てに完全な固定と被覆を提供したい
	- いざとなればネジ6本くらいで全部外せるようにしたい
* フレームがガタガタギシギシ言ってどうにもならない
→[ロックタイトなど](https://www.amazon.co.jp/LOCTITE-ロックタイト-ねじロック-243-中強度タイプ/dp/B01LE9TNQ2)を使う
→いつものネジ止めアルミフレームを止めにする
	- もう少しマシな設計を試みる
	- 外皮にも応力を分散させたいし, 何より格好悪い <- 重要
* わかってたことだが, センサが坂を読んでいる
	- LIDAR とか導入しちゃう?
* タイヤが小さすぎて, 小石にも躓いている
	- 上半身に対して下半身が貧弱すぎる
	- 振動のせいだと思うけれど, odom が壊滅的
	- 最低でもサスほしい
* 制御ゲインが不適切で, スピンに入ると発散して死ぬ

forno, 2016-11-04:
これっていつまでにすることを想定した内容なんですー？

yasu80, 2016-11-04:
@forno 
「する」ことは全く想定せず, 問題とすべきことすべてを洗い出していくために作りました.
そんで, [募集] です

forno, 2016-11-04:
@yasu80 返信有り難うございます。

シミュレートモデルが無いのでそれの追加でしょうか。
ハード班もモデルを作れるようになれば、ハードが注文待ちでもとりあえず全体の確認が取れますので。

私が思いつく他の点はたいていソフト寄りすぎて5号機に限らないので一旦略。必要があれば足します。

yasu80, 2016-11-04:
ハードばっかり書きましたけど, 後々のこと考えると依存関係はっきり書きなおす必要もありますね・・・

forno, 2016-11-04:
依存関係ちゃんと書いてない奴とかあったん？

forno, 2016-11-04:
まあ、何を*どう*使ったのかは残しておいたほうが良いとは思います。

yasu80, 2016-11-06:
@forno 件の三回コンパイルするアレです

yasu80, 2017-01-26:
図は後で作りますが, 重心が低い構造物については高い部分からのモーメントに対して非常に脆弱になりますから, 重心を高く設定したいと思います. いま宣伝用に組み直した奴は二階建てになっています. バッテリーも最低地上高 30 cm 以上に設定したいなというのが感想

yasu80, 2017-03-08:
また改造しましたが, いろいろ問題点が出てます

 ・ フレームが傷んできている
→新しく作りなおすときに, 更新する
 ・ 端子不良が多発
→使い回しをやめて, １度全部新調する
 ・ 運搬のために押しやすくしたい
→取っ手をつける
 ・ ゲイン設定が不適で制限速度が2km/h程度
→調整して最低でも6km/h程度には強化する

yasu80, 2017-03-08:
LRFは増量してもいい. ただしバスパワーコントローラがいる

yasu80, 2017-03-08:
・ USBコントローラが電流不足で落ちている. おそらく @RyodoTanaka  さん の仮説であるが, おそらく正しい. マイコン行きのポートとLRFがバカ食い
→とりあえずは別のポートに繋げばいい
→バスパワーUSBハブを購入

yasu80, 2017-03-09:
・ ずっと目をそらしていたが・・・PCの固定用具が足りない. 俺のVostro君が暴走して吹っ飛んで大怪我だよ
 ・ fifth_robot_launcher 無い fifth_robot_paramの調整をしないと足が遅すぎる.

yasu80, 2017-04-17:
把手は引き紐で代用できることが分かりました

yasu80, 2017-08-09:
何かあったらリオープンします

closed 2017-08-09 by yasu80

forno, 2017-08-09:
直す必要のあるものだけIssue立てて、
referを貼っておくといいかも