CIR-KIT/fifth_robot_pkg #47: 外のmapがダメっぽい原因を知りたい

forno, 2016-11-04:
park_1stの調子を見ると何気によさ気な地図ができている。

対してpark_2ndは正直読めたものでは無い。これは累計旋回角度が関係しているのだろうか？それとも、バックや旋回中の揺れがセンサに響いているのであろうか？

bagファイルが無いと流石に推測も何もできないことに気づきました。
どうにかして欲しいですが、repositoryに込められるとダウンロードの時間がすごい勢いで増えます。回線が1Mbit/sなんです許してください。最悪我慢します()

forno, 2016-11-04:
park_1stのマップについて、もしかして現実の尺度と極端に異なるマップではありませんか？

マップ外側の線が綺麗に出ているように見えたためにしばらく気づきませんでしたが、内側の線がズレて複数回出現しているように見受けられます。
恐らく旋回の際などに直進したと判断されているのではないかと思ってみました。

それとも、斜面が問題を引き起こしているのかもしれません。　できればURGをもう少し上方向に修正して試してみるのも良いかと思います。

yasu80, 2016-11-04:
rosbagでみてみたんですけれど, やたら点群がぴょんぴょんしているのが気になります.
これは, 実際
1 路面状況が石畳のランダムパターンなんでLRFの値(反射強度)がめちゃくちゃに変化しまくる
2 レーザー光線(というよりは機体)がz軸にブレまくっている
ってのがはっきりわかります.

yasu80, 2016-11-04:
記録してあるのが/tf /(LRFの奴) のみなんでodometory とかが出せないでちょっと不便・・・・

forno, 2016-11-04:
まあ、ロボットの移動は読み取れてるので。 tfに吐いてる内容とOdomに吐いてる内容は同じはずですし。

forno, 2016-11-04:
とりあえず頑張って上方向を向けて無視できないか模索するのが最も効果的かと思います。

yasu80, 2016-11-04:
返信ありがとうございます.

それがいいと思いますが, 坂を読んでる分で相当 map が汚れてるんですよね...
※一方上回生は手塗りで補正した模様

yasu80, 2016-11-04:
とにかく, mapが汚いのはご愛嬌として, 話は waypoint_saver をともかくも起動して動かしてやってから, どれくらいの起動ができるかだってことに気が付きました.

最悪 occupancy_grid のパラメータを costmap からいじり倒して, 小さな計測結果を無視できるようにできるはずですし, 今日は車検やらあれやこれやで暇食い過ぎたので明日は (CST, CIR-KIT 標準時による) waypoint 取って, navigation するのが先決でだと気がつく.

forno, 2016-11-04:
あー。ファイル解析してたけど手塗りが早いか

forno, 2016-11-04:
なんかしきい値とかあったから一応白黒じゃなくて値にばらつきでもあるのかと思って調べてたなど。
結論的に言うと、あったけどマジ無意味。ちなみに以下はバイナリの出現文字個数。

```bash
  1980 0000
    770 00cd
   1483 00fe
      1 0a30
      1 0a35
      1 0a78
      1 2030
      1 203a
      1 2070
      1 230a
      1 2e30
      1 2e72
      1 2f6d
      2 3030
      1 3034
      1 3420
      1 3530
      1 3532
      1 3550
      1 4320
      1 4552
      1 524f
      1 5441
      1 5f70
      1 614d
      1 6172
      1 6567
      1 656e
      1 6970
      1 6f74
      1 7063
      1 75022072
    670 cd00
   2611 cdcd
   1112 cdfe
   1601 fe00
   1004 fecd
   4149 fefe
```

forno, 2016-11-04:
とはいえあのマップを見る限り、Amclも死にそうな勢いな気がするので、上に向けるのは検討するようにおねがいしますー。

yasu80, 2016-11-04:
手塗りも早くはなさそうですね・・・
「走行中のエラーパターンを」探してたらしいんで, 二時間以上かかってたぽい

yasu80, 2016-11-04:
センサ位置いじるのが割と建設的ですね. ありがとうございました.

forno, 2016-11-04:
自分の状態を把握するためのセンサが正常に作動しなければシステムは動けないのです。
誤差を取り除くように作れても、仮定から外れた状態から建設的値を出すことはできないのですから。

forno, 2016-11-04:
というかWaypoint generator のruntimeはAmclに依存してるからAmclが正常に動かなければWaypointも取れないぞ。　頑張れ僕らの @yasu80

forno, 2016-11-04:
おっと、更に追記：

Amclのパラメータ変更のプルリクを実機実験前にマージされたので、diffとdiff-correctedの違いがわかっていません。　まあ、センサが飛ぶのであればあまり関係のない話でしょうが、ロボットの位置が飛ぶ場合は戻してみるのがよいかもしれません。

forno, 2016-11-05:
あっと。思い出した。

確かAmclだったか、何かの設定ファイルにLaserScanって書いてあったのを思い出しました。
/most_intenseには変更されてるのにそっちが変わってないのって何か影響しないかなーと思ってみたり。

forno, 2016-11-05:
と、おもったけど/most_intenseはLaserScan型でした。杞憂ですねわかります。
echoesトピックが (sensor_msgs/MultiEchoLaserScan) 型でした。

closed 2016-11-05 by forno

forno, 2016-11-05:
そういえば、装甲をいくらか足したみたいですが、自分自身が見えていたりしませんか？

yasu80, 2016-11-05:
$BFI$a$F$k$C$]$$$1$I8m:9$G;`$s$G$k(B

Outlook for Android<https://aka.ms/ghei36> $B$+$i<hF@(B

$B$=$&$$$($P!"Au9C$r$$$/$i$+B-$7$?$_$?$$$G$9$,!"<+J,<+?H$,8+$($F$$$?$j$7$^$;$s$+!)(B

-

forno, 2016-11-05:
見えてないです(迫真)