CIR-KIT/fifth_robot_pkg #55: USBポートの帯域食いつぶし問題

yasu80, 2017-02-27:
いままでUTM-30LX-EW[www.hokuyo-aut.co.jp/search/single.php?serial=146](メーカ)を使っていたため, これの送ってくるデータは100BASE Ethernet Port 経由で受け取れていました

現状, それを返却しているので, UTM-30LX[url](http://www.hokuyo-aut.co.jp/search/single.php?serial=21)を使っていますが, これはUSBシリアルポート経由でデータを送受信します.

どうやら, USBポートの帯域が埋まって通信ができないようです.
正常に起動し, cmd_vel トピックにリモコンからのコマンドが吐き出されているのですが, これをモータが受け取っていない模様です.

具体的には, レーザを停止するとリモコンから操作ができ, 起動しているとリモコン入力で動いてくれないという状況です.

ただし, scanトピックには通信中常時値が出力され続けているので, おそらく通信自体には成功している模様です.

@tanacchi と私のマシンで微妙に挙動が異なり, 彼のでは urg_node がInvalid error で落ちていますし, 接続が「不安定, たまに Invalid value が帰ってきます.

目下の対策としては,
 ・  転がっているLRFでEthernet経由のやつがないか調べる
 ・ LRFを読むマシンと yp-spur と通信するマシンを adhockモードでネットワーク形成し,分散処理を試みる(達成まで時間がかかるかも)
 ・ バスコントローラが贅沢な感じのマシン持ってる人に動かしてもらう(最悪の方法)
などが考えられます

今後, 検証し報告します.

yasu80, 2017-03-06:
UST-20-LXで確認してみましたが, Invalid valueが見えてます
問題の切り分けに時間がかかりそうです.

forno, 2017-03-06:
Ethernetでもinvalid value になるってことでいいのかしら

yasu80, 2017-03-06:
ちなみに, 相手方のipがデフォルトであることは確認済みです.(192.168.0.10)
なんかこのLRF壊れてるくさいですね

yasu16@yasu16-Vostro-3546:~/Documents/workspace/robots/cirkit5/fifth_robot_pkg/src/fifth_robot_launcher/launch$ ping -c  3 192.168.0.10
PING 192.168.0.10 (192.168.0.10) 56(84) bytes of data.
64 bytes from 192.168.0.10: icmp_seq=1 ttl=64 time=0.147 ms
64 bytes from 192.168.0.10: icmp_seq=2 ttl=64 time=0.151 ms
64 bytes from 192.168.0.10: icmp_seq=3 ttl=64 time=0.158 ms

--- 192.168.0.10 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1998ms
rtt min/avg/max/mdev = 0.147/0.152/0.158/0.004 ms

yasu80, 2017-03-08:
動きましたが, 三個くらい要因があり

 ・ 一個壊れているLRFを踏んづけていた(エラーシグナルを履いていた)
 ・ /dev/ttyACM0を serial_by_idなんとかにしたら動いた
 ・ おそらくUSBコントローラの帯域輻輳ではなく, 電流不足ではないかという @RyodoTanaka 氏のご意見をいただきました

yasu80, 2017-03-08:
RosIpでうごかなかった理由は未確定なので, しばらくあけておきます.
serial_by_idかな?

yasu80, 2017-04-09:
Rounting Table の設定不良であった可能性が高いので, ここは閉じておきます

closed 2017-04-09 by yasu80