CIR-KIT/fifth_robot_pkg #62: モータドライバ T-Frog TF-2MD3-R6 の挙動

yasu80, 2017-05-25:
ずっとトルクの左右比率がおかしかったのを修正しました.

結局原因不明です. いろいろ検証したいのでたてました.

yasu80, 2017-05-25:
[image]
割とまともなmapがとれるとこまできたんですけれど, 実のところハード的にはパラメタファイルを最初にもどしただけで真新しいところが何もない. ひょっとしてマシンの処理能力が悲しみを生んでいたとか?

yasu80, 2017-05-25:
ちなみに60secクオリティです.

forno, 2017-05-26:
先輩が地図作成をするときにバッグファイルと内部時間の再生速度を半分にして地図を作成する技を利用して(おそらく処理時間を稼ぎ)外の地図を作ってることを考えると,
あながち間違っていないのではと思うにゃう

AriYu, 2017-05-26:
PCの処理能力が足りないときやリアルタイムにSLAMする必要がないときは
rosrun gmapping gmapping_replay bagfile_name
を使うといいです。（あんまり覚えてないのであってるか分かりません）
ドキュメントは無かった気がしますが、githubでコードを読めば使い方が分かると思います。

yasu80, 2017-08-08:
諸々の実験を重ねた結果, TF-2MD3-R6 のレベルでは正常に動作していることが推測できるのでひとまずCloseします.

@AriYu 先輩,  @forno 先輩, 貴重なご助言をありがとうございました。

closed 2017-08-08 by yasu80