CIR-KIT/fifth_robot_pkg #64: Projectはログが残らない上、時系列に並ばないので（他の人が）見返しづらいのですにゃ

forno, 2017-08-09:
GithubのProject機能は強力で腰が軽く、素早い行動を可能にしています。

しかしながら、Projectを見ていて思ったことがあります。
URDFとSDFをどちらにするか、迷って決めた要因、決定事項などが
別々のパネルになっていて、さらに別々のカラムにあるために流れがわからないのです。

良い点としては、決定事項だけが並んでいる点などがみやすいです。

そこで、Projectをロードマップとして更に活用するように工夫できないかと思います。
Projectの構成は今のままで、パネル同士に関連をだすために、
関連するIssueを立ち上げてIssue番号を通してはいかがでしょうか。

また、こちらからアドバイスしたいときもどうしていいかわからないので、
やっぱりIssueがほしいなということなのです。

（キャラが意味不明なのは疲れているからです）

yasu80, 2017-08-09:
## Project
 - 並列型の記述ができる
 - トピック分けが便利
 - 記述してある事項をひょいひょい移せる
 ## Issue
 - 単一の話について議論するのに向く
 - 話の流れを保存できる
 - シリアルナンバーがつく

整理してみると, やはりそのように運用するべきだと思います. 特にProjectに貼った話題についてProjectないで議論できないのは痛いので, 決定事項だけ書くようにして迷うことはIssueへ任せようと思います.

さらに工夫するなら, コミット番号への参照でしょうか.
むしろそっちがメインになるくらいの進捗しないと.

ただ. 結局専門のガントチャートツールってわけでもないので, たとえ話題に番号を降って
整理したとしても時系列は番号を追っかけて頭の中で整理する必要があるため
割とわかりにくいものでしょうが無い気もします. Googleカレンダーと同期とかできたらまごうことなく最強なんですけど. (というかそれができたらガントチャートとか要らなくなる...)

Projectでは, 今やるべきことが見えればだいたい良しだと思います.
整理すると

 - Project
   - 決定事項カラム
     - Issueが生える
   - 進捗事項カラム
     - Issueが生える
   -  ログカラム
      - 使い終わったIssueが収まる(番号がつくので一応流れがわかる)

って感じでいきたいとおもいます.

forno, 2017-08-09:
Projectのパネル内にIssue番号はっちゃえー

closed 2017-08-09 by yasu80