CIR-KIT/fifth_robot_pkg #67: gazebo構成デバッグ録

yasu80, 2017-08-11:
# しんどい
http://gazebosim.org/tutorials?tut=ros_roslaunch を参考にしたがこのままではphysicsが動かず死ぬ
 ## rrobotとの比較
 - launch内部
   - robot_descriptionの不足(paramServerにurdfを投げる方法)
   - gazebo_ros 内の urdf_spawner を稼働しないといけない
   
- description系
  - meshes が何者か
  - .gazeboファイルはurdfとキャラかぶって無いか
  - robot_state_publisher が不足

yasu80, 2017-08-11:
``` error
yasu17@yasu17-Inspiron-7460:~/workspace/robots/fifth_robot_pkg/src/fifth_robot_gazebo/launch$ roslaunch fifth_robot_world.launch 
... logging to /home/yasu17/.ros/log/9a502a86-7e8e-11e7-874f-d481d7abf863/roslaunch-yasu17-Inspiron-7460-2860.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yasu17-Inspiron-7460:39157/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [2873]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9a502a86-7e8e-11e7-874f-d481d7abf863
process[rosout-1]: started with pid [2886]
started core service [/rosout]
process[gazebo-2]: started with pid [2889]
process[gazebo_gui-3]: started with pid [2894]
[ INFO] [1502453660.886834951]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1502453660.887580515]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
[gazebo-2] process has died [pid 2889, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/yasu17/workspace/robots/fifth_robot_pkg/src/fifth_robot_gazebo/worlds/fifth_robot.world __name:=gazebo __log:=/home/yasu17/.ros/log/9a502a86-7e8e-11e7-874f-d481d7abf863/gazebo-2.log].
log file: /home/yasu17/.ros/log/9a502a86-7e8e-11e7-874f-d481d7abf863/gazebo-2*.log

```

forno, 2017-08-11:
Path が まちがっている ぞ

yasu80, 2017-08-11:
world -> worlds で解決

closed 2017-08-11 by yasu80