CIR-KIT/fifth_robot_pkg #74: fifth_robot.urdfのパースエラー

yasu80, 2017-09-08:
## 目下調査中

>yasu17@yasu17-Inspiron-7460:~/workspace/robots/fifth_robot_pkg/src/fifth_robot_description$ rosrun >xacro xacro.py fifth_robot.urdf
>xacro: Traditional processing is deprecated. Switch to --inorder processing!
>To check for compatibility of your document, use option --check-order.
>For more infos, see http://wiki.ros.org/xacro#Processing_Order
>XML parsing error: not well-formed (invalid token): line 3, column 2
>when processing file: fifth_robot.urdf

>Check that:
 >- Your XML is well-formed
> - You have the xacro xmlns declaration: xmlns:xacro="http://www.ros.org/wiki/xacro"

c4055d5 (現行バージョン)にて発生を確認

指定箇所にそれらしいもの見えませんし, [もう面倒くさいのでサンプルベースに再構成します](http://bril-tech.blogspot.jp/2016/10/ros4.html)

forno, 2017-09-08:
あなたが叩いてるコマンドの方がわからんでござるよ．

普通は 

```bash
$ rosrun xacro xacro.py fifth_robot.xacro > fifth_robot.urdf
```

みたいな感じでは？

とりあえず間違い無く言えるのは
`rosrun > xacro`
となってる時点で間違いくさいぞ！

forno, 2017-09-08:
ちなみに上のはxacroファイルからurdfをファイルに出力する例を想定した適当なコマンドだけど．
パラメータに吐くにしてもだいたいおんなじ感じのはず．

yasu80, 2017-09-08:
トラブルシューティング板で聞いてきた方法構文エラー検出なんですけども,
どこかの指定(だったような気がする)が指定できた時に解消できました.

forno, 2017-09-08:
xacro通せば元ファイルがurdfだったとしてもきちんとパースエラーチェックはしてくれるけど．
でもどう考えても元のコマンドはおかしい．

forno, 2017-09-08:
まあ，解決してるなら私のツッコミに
「それは違うよ！」
とソース付きレスを返してクローズしてしまってくだち．

yasu80, 2017-09-20:
リダイレクト記号が結局何だったのかわからんです. 貼り付けるときに混入したのかなんなのか・・・

closed 2017-09-20 by yasu80