CIR-KIT/fifth_robot_pkg #78: マルチエコー取得可能なLRFの特性モデルについて

yasu80, 2017-09-13:
現状Gazeboで使用可能なのはHokuyo_Nodeで使われるタイプのUTM-30-LXで,
本番で使いそうなのはUTM-30-LX-EWであるから, マルチエコーモードの有無が大きな争点になりそう.

Gazebo上で導入したい. ただ, 挙動がわからない.

トピック名のリマップはともかくとして, most_intenseとかの機能がどれくらいの差を生むかがわからん. 一応, 現物で確認ができるまで保留?

yasu80, 2017-09-13:
感覚的には光点のばらつきがマシになる気がする.

yasu80, 2017-09-13:
これ読む限りではだいたいそれで良さそう
http://www.roboken.esys.tsukuba.ac.jp/~ohya/pdf/Robomec2012-SAT.pdf
アイデア何かありましたらください()

forno, 2017-09-13:
マルチエコー対応してれば半透明な物体に対応できるって認識でおｋ。

対応してないセンサでそういう物体が中間に入るとソフトウェア工学的には未定義動作ってことになるな。
（もちろん初めに反射したものが見えるけど、半透明は映るか不明なので）

たとえば今までと違ってスリットじゃなくてアクリル板で完全に囲っても動くとかそんな感じじゃないかしら。

yasu80, 2017-09-13:
返答ありがとうございます.

具体的には有意な値がどれくらい得られるかが問題になりますが, 半透明物体とかでうまれる無意味な値(ノイズとして消したい)が入らなくなるので結果的にはマップが綺麗になって返り値が綺麗になりますね.

そうなると, それをどうやってモデル化するかって話にはなるんですけれど, とりあえずはノイズモデルなんかのパラメータを少し弄る程度で良いかと考えています.

本当にノイジーな環境下( eg.金網とその向こうが両方見えてる )においてのモデルを正確に表現したかったら, 一定以上の情報が密集している範囲においてのエコーの挙動をスクリプトにするとかするべきなんでしょうけれど・・・当面はいいかなと.

実のところ実環境で走らしたら, ゴツゴツした縁石とか藪とか, 綺麗な反射面のない物体に対してなんかこうクソな感じの値が帰ってきて泣けるので均すか他の何かで補佐するしたい感じなんですけれどそれはまた別で考えます

closed 2017-09-13 by yasu80