CIR-KIT/fifth_robot_pkg #81: YPSpur落ちる問題について質問

RyodoTanaka, 2017-09-20:
5号機を開発中の皆さま，YPSpurについて質問があります．
現在，防衛大(以下NDA)の方がYPSpurを使って開発を行っているそうなのですが，
- Ethernet タイプのLRF x2
- YPSpur

を起動すると，しばらくたってからエラーで落ちるそうです．
もし，何か似たような症状や解決策があれば教えていただけると嬉しいです．
すでに #55 の電力不足の可能性も考えましたが，EthernetタイプのLRFであり，電源をUSBバス以外から取っているため，これが原因ではないように思います...
下記がエラー発生時の画面だそうです．
[image]
よろしくお願いします！

RyodoTanaka, 2017-09-29:
@yasu80 
回答ありがとうございます．
あと，返事遅くなってすみません．
先方で解決したそうです．
以下，原因と解決法です．
# 原因
YPSpurと駆動モーターの電源が同一であり，モーターのノイズがYPSpurに流れ込んでいたため

# 解決方法
YPSpur側の電源とモーター側にそれぞれコンデンサをぶち込む．
5号機でも起き得ると思うので，同様の問題があれば助けになれば嬉しいです．

yasu80, 2017-09-29:
@RyodoTanaka 
お知らせありがとうございます.
解決した様子なのでCloseします

closed 2017-09-29 by yasu80