CIR-KIT/fifth_robot_pkg #85: GPSのテスト

yasu80, 2017-10-02:
Hemesphere A-100というGPSモジュールがあります.
消費電力が2Wしかないので使ってみたいですが, 暇なROS使い居ませんか...?

nmea形式のバイナリがぺろぺろ流れてくるのでそれをnmea_gps_driverでひっつかんで, odomをもらってきます.
起動は12Vをくれてやればいいとのこと.

http://wiki.ros.org/nmea_gps_driver
http://wiki.ros.org/gps_common
http://www.hemgps.com/a100.html
http://www.hemgps.com/dl/CrescentA100catalogjp.pdf

Kose-i, 2017-10-02:
こっちのパッケージを使うことが推奨されていました.

http://wiki.ros.org/nmea_navsat_driver

Kose-i, 2017-10-02:
マニュアルです

http://www.hemgps.com/dl/A100_manual_jp.pdf

Kose-i, 2017-10-06:
`rosrun nmea_navsat_driver nmea_serial_driver` (初期は4,800で設定されている)
`rosparam set /nmea_serial_driver/baud [数字]` (GPS側のbaudrate値と一致させる必要あり)

　　設定可能な値は：4,800/9,600/19,200/38,400/57,600/115,200 bpsです (現在は19,200で設定すると通信ができるようになります)

(一度、baudが一致すると変更された値でも読み込むことがあるようです.また、一致していない値で、
`rostopic echo /fix`
すると、値を本来通信可能な19200と設定しても読み込まないことがあるため、その時はC-cでnmea_navsat_driverを止めてください)

`rosparam set /nmea_serial_driver/port [/dev/~]`(ケーブルを刺した時に現れるデバイスです)
`rostopic echo /fix`

<hr>

--確認用-------

`rosparam get /nmea_serial_driver/baud`
`rosparam get /nmea_serial_driver/port`

yasu80, 2017-10-15:
レポありがとうございます.
少ししたら総括しますね

yasu80, 2017-10-19:
GPS導入は先送りか現地実装とします.
使えることだけ確認できたのでClose

closed 2017-10-19 by yasu80