CIR-KIT/fifth_robot_pkg #93: TF2-MD6(yp-spur)からの不正パケット送信

yasu80, 2017-10-19:
@marubashi11 の環境で発生,
不正パケットの送信があったとコンソール出力があり ,以後唐突に終了する.
同一環境にて私のマシン( 4.4.0-97-generic)では発生せず. Linuxカーネル4.11系環境下(Ubuntu-hwe-edge)にて発生.

https://github.com/open-rdc/orne_navigation/issues/14 にも

>公式ページで情報を確認したところ、LinuxカーネルのバージョンによってはFTDIチップのドライバにバグがあり、通信ができなくなるということがわかりました。

との報告がみられるが, 出典がないため詳細不明.

エラーログをください.

marubashi11, 2017-10-19:
[ERROR] [1508385744.218176146]: Couldn't open joystick /dev/input/js0. Will retry every second.
[openUSB-2] process has died [pid 22676, exit code 1, cmd /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/fifth_robot_pkgs/fifth_robot_launcher/shell/openUSB.sh __name:=openUSB __log:=/home/marubashi/.ros/log/4fde6164-b482-11e7-9a7d-a8137493dc16/openUSB-2.log].
log file: /home/marubashi/.ros/log/4fde6164-b482-11e7-9a7d-a8137493dc16/openUSB-2*.log
[ypspur_coordinator-3] process has died [pid 22678, exit code 127, cmd /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/third_party/ypspur_ros_bridge/scripts/ypspur-coordinator.sh -p /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/fifth_robot_pkgs/fifth_robot_launcher/ypspur_params/fifth_robot.param -d /dev/serial/by-id/usb-T-frog_project_T-frog_Driver-if00 __name:=ypspur_coordinator __log:=/home/marubashi/.ros/log/4fde6164-b482-11e7-9a7d-a8137493dc16/ypspur_coordinator-3.log].
log file: /home/marubashi/.ros/log/4fde6164-b482-11e7-9a7d-a8137493dc16/ypspur_coordinator-3*.log

teleop.launchを実行しようとすると、まずこのエラーが出て不正パケットの送信エラーまで行きません。
このエラーを回避するための手順を忘れてしまったのですが、もう一度教えていただけますか？

yasu80, 2017-10-19:
@marubashi11
chmod 777 /dev/serial/by-id/（デバイス）

又はシェルで同じこと

たしかエラーコードをホームに保存した

marubashi11, 2017-10-19:
[error (copy).txt](https://github.com/CIR-KIT/fifth_robot_pkg/files/1397305/error.copy.txt)

Please tell me if this link can't open.
(Sorry, now I can't enter Japanese.)

marubashi11, 2017-10-19:
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ ls
build  devel  src
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ source devel/setup.bash 
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ source devel/setup.bash 
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ rosrun fifth_robot_launcher openUSB.sh 
chmod: cannot access '/dev/serial/by-id/usb-T-frog_project_T-frog_Driver-if00': No such file or directory
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ rosrun fifth_robot_launcher openUSB.sh 
marubashi@ubuntu-CFSZ6-1L:~/fifth_ws$ roslaunch fifth_robot_launcher teleop.launch 
... logging to /home/marubashi/.ros/log/311b9ea4-b4a5-11e7-a453-a8137493dc16/roslaunch-ubuntu-CFSZ6-1L-18332.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-CFSZ6-1L:34661/

SUMMARY
========

PARAMETERS
 * /joy_node/autorepeat_rate: 20
 * /joy_node/deadzone: 0.3
 * /joy_node/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /teleop_twist_joy/axis_angular: 0
 * /teleop_twist_joy/axis_linear: 1
 * /teleop_twist_joy/enable_button: 8
 * /teleop_twist_joy/enable_turbo_button: 10
 * /teleop_twist_joy/scale_angular: 1.0
 * /teleop_twist_joy/scale_linear: 1.0
 * /teleop_twist_joy/scale_linear_turbo: 2.0

NODES
  /
    joy_node (joy/joy_node)
    openUSB (fifth_robot_launcher/openUSB.sh)
    teleop_master_node (teleop_master/teleop_master_node)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
    ypspur_coordinator (ypspur_ros_bridge/ypspur-coordinator.sh)
    ypspur_ros_bridge (ypspur_ros_bridge/wait_and_ypspur_ros_bridge.sh)

auto-starting new master
process[master]: started with pid [18343]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 311b9ea4-b4a5-11e7-a453-a8137493dc16
process[rosout-1]: started with pid [18356]
started core service [/rosout]
process[openUSB-2]: started with pid [18367]
sudo: no tty present and no askpass program specified
process[ypspur_coordinator-3]: started with pid [18375]
++++++++++++++++++++++++++++++++++++++++++++++++++
YamabicoProject-Spur
 Ver. 1.14.0
++++++++++++++++++++++++++++++++++++++++++++++++++
    with SSM

 SSM is not available.
Device Information
 Port    : /dev/serial/by-id/usb-T-frog_project_T-frog_Driver-if00 
process[ypspur_ros_bridge-4]: started with pid [18377]
process[teleop_master_node-5]: started with pid [18379]
process[joy_node-6]: started with pid [18380]
process[teleop_twist_joy-7]: started with pid [18388]
[openUSB-2] process has died [pid 18367, exit code 1, cmd /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/fifth_robot_pkgs/fifth_robot_launcher/shell/openUSB.sh __name:=openUSB __log:=/home/marubashi/.ros/log/311b9ea4-b4a5-11e7-a453-a8137493dc16/openUSB-2.log].
log file: /home/marubashi/.ros/log/311b9ea4-b4a5-11e7-a453-a8137493dc16/openUSB-2*.log
Warn: Baudrate setting is not supported on this device.
Applying parameters.
YP-Spur coordinator started.
Warn: Illegal packet 'SS038400' received.
[... 91 lines cut ...]
Warn: Illegal packet 'DPGmP' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'E`D@@@@@' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'DpG|Kw@' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'EpD@BR_@' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'Ep@@BR_@' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'DPCmP' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Warn: Illegal packet 'F@@@@@FP' received.
Warn: Illegal packet '0Ee' received.
Warn: Illegal packet '' received.
Connection to /dev/serial/by-id/usb-T-frog_project_T-frog_Driver-if00 was closed.
Trajectory control loop stopped.
[ypspur_coordinator-3] process has died [pid 18375, exit code 1, cmd /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/third_party/ypspur_ros_bridge/scripts/ypspur-coordinator.sh -p /home/marubashi/fifth_ws/src/fifth_robot_pkg/src/fifth_robot_pkgs/fifth_robot_launcher/ypspur_params/fifth_robot.param -d /dev/serial/by-id/usb-T-frog_project_T-frog_Driver-if00 __name:=ypspur_coordinator __log:=/home/marubashi/.ros/log/311b9ea4-b4a5-11e7-a453-a8137493dc16/ypspur_coordinator-3.log].
log file: /home/marubashi/.ros/log/311b9ea4-b4a5-11e7-a453-a8137493dc16/ypspur_coordinator-3*.log

yasu80, 2017-10-19:
今から低レイヤハッキングになるのはNGなのですみませんが代機でいきましょう.
私のVostroと @dorayaki0903yamamoto  のLavieなら動くはずです.

closed 2017-10-19 by yasu80

yasu80, 2017-10-21:
xenial（Ubuntu16）系列4.9以降に存在するカーネルのFDTIドライバ周辺バグですが, 
Grubの
Advanced setting for ubuntu
から旧バージョンカーネルによる起動を試みることにより回避できます（ダウングレードやリインストールは必要ない）