CIR-KIT/fifth_robot_pkg #95: yaml_reader リメイク案

yasu80, 2017-10-19:
とりあえず基幹機能だけ

## Input

 - waypoint.yaml
   - latest
     - essential
       - x
       - y
       - type (default:passing)
       - level_alert (max:=5, min:=1, default:3)
       - level_acculate (float, radius, default=???)
     - optional
       - name
       - flag
 
 - request_next
 - request_prev
 
 ## OutPut

 - waypoint (~Service?~ Topic, need dynamic reconfigure on costmap or something)
   - x
   - y
   - type
   - level_alert
   - level_accurate
   - name
   - flag
   
   
  とりあえずこんなかんじ, できたら複数のwaypointを指定個数表示して経路表示をいい感じにできないかなーとか思わんでもない

forno, 2017-10-30:
回転角度が足りてないのでは？

forno, 2017-10-31:
yaml readerが食べるファイル仕様案

```yaml
- position:
    x: 0
    y: 0
    z: 0
  orientation:
    x: 0
    y: 0
    z: 0
    w: 1
  radius: 1
  name: "point name" # optional

yasu80, 2017-10-31:
そのような仕様になると思われます.

yasu80, 2017-11-23:
#96 参照

closed 2017-11-23 by yasu80