CIR-KIT/fifth_robot_pkg #97: 走行中に local_costmap が更新されていない

tanacchi, 2017-10-21:
[image]
rviz 上で local_costmap を表示させている様子です。
地図上で検知された obstacles の周辺には水色の点群がないのがわかります。しかしレーザーから読み取れるはずの obstacles に関しては反応がありません。
また、goal の指定した時点で検出された obstacle に関しては （local_costmap に反映されていないものの）避けて進んでくれていますが、goal を指定した後ではそれができていません。
現在 wiki 等で解決を急いでいます。何かアイデアがある方はお願いします。

tanacchi, 2017-10-21:
中央右の obstacle の内部に点群があるのも謎ですが。

yasu80, 2017-10-21:
参考?

https://qiita.com/MoriKen/items/1f1f2d1e6ef0046ec12a

tanacchi, 2017-10-29:
先輩方かた頂いたアドヴァイス
* そもそも publish & subscribe が上手くいっているのか rqt_graph などで確認する
* namespace の確認もしておく
* topic の型は正しく設定されているのか確認

param は散々見てきたけどまだ勉強不足な点もあると思うので引き続き行うつもり
rqt_graph での確認を指定なかったので実機の配線等が済み次第、複数LRFのテストとともに検証

tanacchi, 2017-10-29:
気になったことメモ
* [ ] third_robot の base_local_planner の TebLocalPlannerROS: が気になる
* [ ] fifthの common_params の laser_frame が気になる

tanacchi, 2017-10-29:
てか十中八九
↑のふたつ目かと

yasu80, 2017-10-30:
common_paramsの件,

>This line sets parameters on a sensor mentioned in observation_sources, and this example defines laser_scan_sensor as an example. The "frame_name" parameter should be set to the name of the coordinate frame of the sensor, the "data_type" parameter should be set to LaserScan or PointCloud depending on which message the topic uses, and the "topic_name" should be set to the name of the topic that the sensor publishes data on. The "marking" and "clearing" parameters determine whether the sensor will be used to add obstacle information to the costmap, clear obstacle information from the costmap, or do both.

これっぽいですね

Kose-i, 2017-10-30:
>  fifthの common_params の laser_frame が気になる（ base_laser -> laser???）

対応してみました。

kousei-costmapブランチではマスターから

- /home/tamura-kosei/robots/fifth_robot_pkg/src/fifth_robot_description/map/gazebo_willowgarage.pgm
- /home/tamura-kosei/robots/fifth_robot_pkg/src/fifth_robot_description/map/gazebo_willowgarage.yaml

のマップを追加し、

- /home/tamura-kosei/robots/fifth_robot_pkg/src/fifth_robot_pkgs/fifth_robot_2dnav/costmap/costmap_common_params.yaml

を指摘された付近について調査して修正しました。

`scan_frame` を `hokuyo_link`へ、 `topic` を `scan`に変更したところ正常にlocal costmapが更新されるようになりました。

1c7c0e5d4b4ef33c239439524e6fe5400cef5c32 にて確認できます。

tanacchi, 2017-10-30:
ありがとうございました。電装が完成され次第、実機でのテストも行います。

forno, 2017-10-30:
ひとつ忠告とリマインドですが．
fifth_robot_2dnav パッケージは共用扱いなので
今回の修正を受け入れることはできません．

実機は元の設定の方が動くはずです．

tanacchi, 2017-10-30:
ありがとうございます。承知しております。
`topic` は `most_intense` のままで、 `scan_frame` を `laser` に変更して一旦テストしてみます。
実際はLRF ２個持ちなのでそこでまた設定しなおすつもりです。

yasu80, 2017-11-23:
ira_laser_toolsに関する不具合と切り分けたいので, とりあえず改善できたlocal_costmapについては(当該issueについては)closeしたいです

closed 2017-11-23 by yasu80