CIR-KIT/fifth_robot_pkg #99: Gazeboのオドメトリ向上

yasu80, 2017-10-26:
色々試しましたが Odometory がダメすぎてamclが可愛そうです.

 - controller.yaml内wheel_separationが不適

おそらくfifth_robot_urdf内の値に当てはめると, Wheel_Separationはjoint間の距離では_なく_
車輪の幅も含んだものであると推察されます. これだけで大分Odomが改善しました.
下は一回転した際の状況です.

[image]

とはいえまだ誤差が無視できないので(amclの最尤推定を妨げるため)何か調整箇所があったら報告ください.

yasu80, 2017-10-26:
willowあたりの重いテクスチャ出してると処理落ちでTFのジャンプ検出のエラーメッセージが出ますが, これについては端末の処理能力によると思われるので無視します(?)

しかし, 端末ごとに挙動が異なることも事実なので個体合わせも必要なのかな...?

yasu80, 2017-10-31:
なんだか下火なのでCloseでいいですよね

closed 2017-10-31 by yasu80

forno, 2017-10-31:
下火というか，実用に耐えうるだけのパラメータになったというか

yasu80, 2017-10-31:
実際のところ不具合の原因も, 個体差の実態もよくわかっていませんから実質未解決です
ですが, 今はそれどころじゃないのでCloseしました.

forno, 2017-10-31:
依然として存在するのね．了解ですー