CIR-KIT/fifth_robot_pkg #112: TF2-MD6モータドライバとUbuntu16.04 (Linux 4.4.9以降)通信時に発生する不正パケット送信問題

yasu80, 2018-05-17:
標記の件について，未解決となっておりましたが，
未だに多数の環境にて発生しており原因の特定に至っておりません．

以下のデータについてカッコ内のコマンドを用いて調べ，提供おねがいします．
・機種
・Ubuntu version および linuxカーネルバージョン(uname -a)
・USBバスコントローラの種類(lspci)

直接は関係ありませんが，カーネルバージョンによってはNIC系列に以上がある場合も認められ，実用に支障をきたす場合があるので併記いただきたいです．

通信可能．不正パケット発生せず．
・機種
ThinkPad X230 
・Ubuntu version および linuxカーネルバージョン(uname -a)
Linux yasu-ThinkPad-X230 4.4.0-124-generic #148-Ubuntu SMP Wed May 2 13:00:18 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
・USBバスコントローラの種類(lspci)
pkgs$ lspci
00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C210 Series Chipset Family MEI Controller #1 (rev 04)
00:16.3 Serial controller: Intel Corporation 7 Series/C210 Series Chipset Family KT Controller (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C210 Series Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.1 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 2 (rev c4)
00:1c.2 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation QM77 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C210 Series Chipset Family SMBus Controller (rev 04)
02:00.0 System peripheral: Ricoh Co Ltd PCIe SDXC/MMC Host Controller (rev 07)
03:00.0 Network controller: Intel Corporation Centrino Advanced-N 6205 [Taylor Peak] (rev 34)

Kose-i, 2018-06-11:
`git clone openspur/yp-spur` で再度おとしてきて、`make install` することでうまく行きました。

yasu80, 2018-06-11:
報告ありがとうございます（´・ω・｀）
一応原因不明ですが解決したということでCloseします

closed 2018-06-11 by yasu80