CIR-KIT/fifth_robot_pkg #140: 3D Mapping の負荷について

tanacchi, 2018-10-28:
### マシンに求められる性能
* ストレージに 50GB 以上の余裕があること
* メモリ 8GB 以上

### 現状発生しているトラブル
@tanacchi の PC ではマップ作成時にメモリが溢れて死ぬ

### 打開策　＋（問題点）
1. Nodelet を用いてリソース節約（やり方をまだ知らない）
2. 1年生の高スペック PC を用いる（最悪 Windows 飛ばすことになる）
3. TX2 を利用する（他の機器との接続が確認できていない。ストレージ足りない＆外付けでは速度が足りない）

yasu80, 2018-10-28:
メモリの増設が手っ取り早いのでは？だいたい4000円くらいです。これくらいなら折半してもいいような。

tanacchi, 2018-10-28:
1年生マシンを使わせてもらえることになりました。

yasu80, 2018-10-28:
一回生のマシンもメモリが8GB入ってるとは考えにくいのですが…

tanacchi, 2018-10-28:
入　っ　て　ま　し　た

tanacchi, 2018-10-28:
ただ、８GBあったところで足りる保証がないので
1. `blam_slam` が `/blam/blam_slam/octree_map` を Publish するのをやめさせる
2. `/blam/blam_slam/octree_map_update` (PointCloud2) を 各々 pcd ファイルに保存
3. （多分自分が書くことになるんだろうけど）そのファイルを統合し1つの pcd ファイルとして保存する

という工程を踏めば確実かと思われます。

yasu80, 2018-10-28:
人権ガン積みくそつよマッシーンで殴る作戦、了解しました
うひゃひゃ～

動作推奨環境で動かないってことも考えづらいのですが…了解です

tanacchi, 2018-10-29:
Map の resolution (粒度) を低くすることで対処できそうです。
一応 pcd ファイルを統合する方法はわかりましたが
`/blam/blam_slam/octree_map` はどのみち Publsh されるので一緒。（Publish させないように変更する手もあるけど気が狂いそう）

tanacchi, 2018-10-29:
/blam/blam_slam/octree_map を Publsh させないようにしましたが
リソースを食いまくるのに変わりはありませんでした。

粒度を落として自己位置推定が機能するかの検証が必要です

tanacchi, 2018-10-29:
マップの粒度を落とす方向で行くことにしましたので閉じます。

tanacchi, 2018-10-29:
> /blam/blam_slam/octree_map を Publsh さ> せないようにしましたが
> リソースを食いまくるのに変わりはありませんでした。

訂正します。
データを生成させないようにするのを忘れました

tanacchi, 2018-10-29:
結局、ソリューションとして2通りあります

* マップの粒度を減らす
* 細切れにして統合する

tanacchi, 2018-10-29:
一応 pcd ファイル統合ツールはできつつありますが、
結局読みだすタイミングでメモリが圧死してるので
やはり粒度を減らす方向が妥当かと思います。

tanacchi, 2018-10-30:
マップ作成時に注意することですが
僕のマシンだけの話かもしれませんけど
普通に bag を再生してしまうと、
自己位置推定のための処理が間に合わないのか知りませんが、破綻します。

また map の resolution を半分にして、
bag 再生のレートを半分にすると（処理時間が２倍）
同程度の粒度の map が生成されるようです。

自己位置推定の精度は圧倒的にこのやり方が良いです。
メモリの消費量は若干こっちの方が少ないか？

要するに
* resolution を必要最低限に下げる
* bag 再生のレートを可能な限り下げる
といけそう

tanacchi, 2018-10-30:
resolution :      1.5
rosbag rate :   0.5

で何とか死なずに、かつ自己位置推定も機能しました。雑木林で自己位置がズレる模様
もう少し点群の密度を上げたいところ。
1年生マシンならうまくいくかしら

tanacchi, 2018-10-30:
マップを細切れにする手法は、update される点群はそれまでに蓄積されたマップデータをもとに生成しているのがソースコードからわかりましたのでボツです。

tanacchi, 2018-11-07:
@syu-kwsk 君のPCならある程度の粒度でマップ生成が可能とわかりました。
結局、
* resolution を下げる
* bag 再生のレートを可能な限り下げる
というのが現状での解決策となります。

閉じます。

closed 2018-11-07 by tanacchi